FreeOpcUa/opcua-asyncio #1716: Error "list index out of range" when calling method in server-methods.py

harriv, 2024-10-03:
**Describe the bug** <br />

When calling mulptily method in examples/server-methods.py from OPC Router (https://opc-router.com), following error message is created:

```
INFO:asyncua.server.uaprocessor:Read request (User(role=<UserRole.User: 3>, name=None))
DEBUG:asyncua.server.binary_server_asyncio:_process_received_message 57 57
DEBUG:asyncua.server.uaprocessor:process_message NodeId(Identifier=554, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>) RequestHeader(AuthenticationToken=NodeId(Identifier=1005, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>), Timestamp=datetime.datetime(2024, 10, 3, 11, 47, 20, 83604, tzinfo=datetime.timezone.utc), RequestHandle=22, ReturnDiagnostics=0, AuditEntryId=None, TimeoutHint=0, AdditionalHeader=ExtensionObject(TypeId=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>), Body=None))
INFO:asyncua.server.uaprocessor:translate browsepaths to nodeids request (User(role=<UserRole.User: 3>, name=None))
ERROR:asyncua.server.uaprocessor:Error while processing message
Traceback (most recent call last):
  File "D:\Projektit\test\OMA_test\venv\Lib\site-packages\asyncua\server\uaprocessor.py", line 143, in process_message
    return await self._process_message(typeid, requesthdr, seqhdr, body)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Projektit\test\OMA_test\venv\Lib\site-packages\asyncua\server\uaprocessor.py", line 319, in _process_message
    paths = await self.session.translate_browsepaths_to_nodeids(params.BrowsePaths)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Projektit\test\OMA_test\venv\Lib\site-packages\asyncua\server\internal_session.py", line 184, in translate_browsepaths_to_nodeids
    return self.iserver.view_service.translate_browsepaths_to_nodeids(params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Projektit\test\OMA_test\venv\Lib\site-packages\asyncua\server\address_space.py", line 188, in translate_browsepaths_to_nodeids
    results.append(self._translate_browsepath_to_nodeid(path))
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Projektit\test\OMA_test\venv\Lib\site-packages\asyncua\server\address_space.py", line 194, in _translate_browsepath_to_nodeid
    if not path.RelativePath.Elements[-1].TargetName:
           ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^
IndexError: list index out of range

```
UaExpert works fine with this method, so there might be also implementation problem in the OPC Router, but I'm not familiar enough with the library and protocol to tell what is happening.

**To Reproduce**<br />
Steps to reproduce the behavior incl code:

1. Run server-methods.py
2. Call mulptiply method from OPC Router

**Expected behavior**<br />
A clear and concise description of what you expected to happen.

Method returns the correct value with out no errors.

**Version**<br />
Python-Version:  3,12.5 <br />
opcua-asyncio Version (e.g. master branch, 0.9): 1.1.5

harriv, 2024-10-03:
I took a look with WireShark, OPC Router does TranslateBrowsePathsToNodeIdsRequest, which UaExpert doens't do:

[image]

[image]

[image]

harriv, 2024-10-29:
I took another look at this. 

Asyncua expects the relative path to have some content:

[image]

But the query actually has zero elements in relative path:

[image]

It is TranslateBrowsePathsToNodeIds which fails:  https://reference.opcfoundation.org/Core/Part4/v104/docs/5.8.4

harriv, 2024-10-29:
Investigating further, based on table 48 here: https://reference.opcfoundation.org/Core/Part4/v104/docs/5.8.4 - BadNothingToDo should be returned if relativePath is empty list. Adding this however didn't solve the problem.

Tested also with Prosys OPC UA Simulation server, it worked fine.

schroeder-, 2024-10-30:
relativePath can't be empty. 
From the specs:
The path to follow from the startingNode.The last element in the relativePath shall always have a targetName specified. This further restricts the definition of the RelativePath type. The Server shall return Bad_BrowseNameInvalid if the targetName is missing.The RelativePath structure is defined in 7.26.

So we should return Bad_BrowseNameInvalid.

schroeder-, 2024-10-30:
Ok BadNothingToDo is the result. This is a bit misleading in the spec.