FreeOpcUa/opcua-asyncio #1744: Use http instead of tcp

HerrMuellerluedenscheid, 2024-11-11:
I took the very basic server example and changed the protocol from `opc.tcp` to `opc.http`. I want to deploy the server to heroku but that apparently does not support tcp endpoints and requires http/s.

This is the server:
```python

import asyncio

from asyncua import Server

async def main():
    server = Server()
    await server.init()
    server.set_endpoint('opc.http://0.0.0.0:4840/')
    uri = "http://examples.freeopcua.github.io"
    idx = await server.register_namespace(uri)

    myobj = await server.nodes.objects.add_object(idx, "MyObject")
    myvar = await myobj.add_variable(idx, "MyVariable", 6.7)
    await myvar.set_writable()
    
    async with server:
        while True:
            await asyncio.sleep(1)
            new_val = await myvar.get_value() + 0.1
            await myvar.write_value(new_val)

if __name__ == '__main__':
    print('starting server')
    asyncio.run(main())

```
When connecting with a basic client:
```python
from asyncua import Client
url = "opc.http://0.0.0.0:4840/freeopcua/server/"
async with Client(url=url) as client:
    _logger.info("Root node is: %r", client.nodes.root)
```
I'm getting this error:
```
[...]

INFO:asyncua.client.client:find_endpoint [EndpointDescription(EndpointUrl='opc.http://0.0.0.0:4840/', Server=ApplicationDescription(ApplicationUri='urn:freeopcua:python:server', ProductUri='urn:freeopcua.github.io:python:server', ApplicationName=LocalizedText(Locale=None, Text='FreeOpcUa Python Server'), ApplicationType_=<ApplicationType.ClientAndServer: 2>, GatewayServerUri=None, DiscoveryProfileUri=None, DiscoveryUrls=['opc.http://0.0.0.0:4840/']), ServerCertificate=None, SecurityMode=<MessageSecurityMode.None_: 1>, SecurityPolicyUri='http://opcfoundation.org/UA/SecurityPolicy#None', UserIdentityTokens=[UserTokenPolicy(PolicyId='anonymous', TokenType=<UserTokenType.Anonymous: 0>, IssuedTokenType=None, IssuerEndpointUrl=None, SecurityPolicyUri=None), UserTokenPolicy(PolicyId='certificate_basic256sha256', TokenType=<UserTokenType.Certificate: 2>, IssuedTokenType=None, IssuerEndpointUrl=None, SecurityPolicyUri=None), UserTokenPolicy(PolicyId='username', TokenType=<UserTokenType.UserName: 1>, IssuedTokenType=None, IssuerEndpointUrl=None, SecurityPolicyUri=None)], TransportProfileUri='http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary', SecurityLevel=0)] <MessageSecurityMode.None_: 1> 'http://opcfoundation.org/UA/SecurityPolicy#None'
INFO:asyncua.client.ua_client.UASocketProtocol:close_secure_channel
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection
Traceback (most recent call last):
  File "cloud_demo_app/client/main.py", line 76, in <module>
    asyncio.run(main())
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "cloud_demo_app/client/main.py", line 26, in main
    async with Client(url=url) as client:
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/asyncua/client/client.py", line 95, in __aenter__
    await self.connect()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/asyncua/client/client.py", line 311, in connect
    await self.create_session()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/asyncua/client/client.py", line 511, in create_session
    ep = Client.find_endpoint(response.ServerEndpoints, self.security_policy.Mode, self.security_policy.URI)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/asyncua/client/client.py", line 130, in find_endpoint
    raise ua.UaError(f"No matching endpoints: {security_mode}, {policy_uri}")
asyncua.ua.uaerrors._base.UaError: No matching endpoints: 1, http://opcfoundation.org/UA/SecurityPolicy#None
```
Do you have a hint how to work with http over tcp?

schroeder-, 2024-11-11:
Not supported currently

oroulet, 2024-11-11:
and really far from support. the https requires serialiazation to something else (json if I remember correctly) than binary