FreeOpcUa/opcua-asyncio #1758: Nodeset exporter should include the full namespace array, not just the namespaces of the nodes that are exported

CraigBuilds, 2024-12-05:
**Describe the bug** <br />
The namespace index of exported nodes will not match up with what is observed in UaExpert. This is because the nodeset only contains the namespaces of nodes in the NodeSet. 

**To Reproduce**<br />
Connect to a server and export the XML using a subset of all nodes, e.g
```python
    client = Client(url)
    print("Connecting to server...")
    async with client as client:
        print("Connected to server")
        objects_node = client.get_objects_node()
        all_nodes: List[Node] = []
        await recursively_populate_children(objects_node, all_nodes)
        filtered_nodes: List[Node] = []
        for node in all_nodes:
            if await is_in_namespace(node, ns=3):
                filtered_nodes.append(node)
        await client.export_xml(filtered_nodes, output_file_path)
```
There will only be one element in the `NamespaceUris` array, e.g. 
```xml
  <NamespaceUris>
    <Uri>http://www.helloworld.com/OPCUA/MyNamespace</Uri>
  </NamespaceUris>
```
And the namespace index of nodes will all be `1`, and not `3` like they should be. (I think they are not `0`, because the `NamespaceUris` implicitly contains `http://opcfoundation.org/UA/` as index `0`, even if it looks like the `0` index is `http://www.helloworld.com/OPCUA/MyNamespace`.

**Version**<br />
Python-Version: 3
opcua-asyncio Version: 1.1.5

CraigBuilds, 2024-12-05:
I have used this as a workaround:

This calls `add_namespace_uri_els` with all namespaces before calling `build_etree`. To get around the issue of the use of the `_addr_idx_to_xml_idx` map while building the etree for each node, it overrides the `_make_idx_dict` function so the address space idxs are the same as the etree idxs. 

```python
async def export_xml_with_all_namespaces(client: Client, nodes: List[Node], output_file_path: str, export_values: bool):
        print("Warning! Exporting all namespaces is an experimental feature and may not work as expected.")
        exp = XmlExporter(client, export_values=export_values)
        ns_array = await exp.server.get_namespace_array()
        exp._add_namespace_uri_els(ns_array)
        def _make_idx_dict(idxs: List[int], ns_array: List[str])-> Dict[int, int]:
            return {idx: idx for idx in idxs}
        exp._make_idx_dict = _make_idx_dict
        await exp.build_etree(nodes)
        await exp.write_xml(output_file_path)
```

chandryan, 2025-04-26:
Hi,
the behavior you observe is expected because namespace indices in OPC UA are not fixed - they *must not* be fixed. Think about a server loading multiple XML files, how would you make sure the namespace indices don't collide?

You need to look up the index based on the URI of the namespace you need. From the [OPC UA spec](https://reference.opcfoundation.org/Core/Part5/v104/docs/6.3) note that only index 0 and 1 are reserved and the only other guarantee you get is that the indices don't change while you're connected:

> Index 0 is reserved for the OPC UA namespace, and index 1 is reserved for the local [Server](https://reference.opcfoundation.org/search/21?t=Server). Clients may read the entire [namespace table](https://reference.opcfoundation.org/search/21?t=namespace%20table) or they may read individual entries in the [namespace table](https://reference.opcfoundation.org/search/21?t=namespace%20table). The [Server](https://reference.opcfoundation.org/search/21?t=Server) shall not modify or delete entries of the [namespace table](https://reference.opcfoundation.org/search/21?t=namespace%20table) while any client has an open session to the [Server](https://reference.opcfoundation.org/search/21?t=Server), because clients may cache the [namespace table](https://reference.opcfoundation.org/search/21?t=namespace%20table)

Even with only one NodeSet2 XML file present the indices in that file are not the same as the ones you would see when a server is running, since in a NodeSet 2 XML there is no local Server namespace. The indices only make sense in the scope of the XML file you're looking at:

- ns0= is the OPC UA namespace, it's always present and not explicitly mentioned in '<NamespaceUris>'
- ns=1 will correspond to the first namespace you see in '<NamespaceUris>', 2 the second and so on

oroulet, 2025-04-26:
Yes correct. This is per design