FreeOpcUa/opcua-asyncio #1766: How-to rename object on running opc ua server?

PetroGolovnya, 2024-12-23:
**Describe the bug** <br />
I trying to rename object on running opc ua server.  
From old DisplayName is 'SNMP' to new DisplayName is 'SNMP changed'.
But object name in tree not changed. 
Code received new name from redis subscription. 

**To Reproduce**<br />
Here is code example:
`async def object_changed(jval):
    org_id = jval["args"]["id"]
    query = "SELECT po.id_p, po.address, p.name FROM page_object AS po LEFT JOIN page AS p ON "
    query += f"p.id_p=po.id_p WHERE po.id_p={org_id}"
    org, = await db.select(query, f"Database error on object {org_id}:")
       
    node_ = server.get_node(ua.NodeId(org_id, 1))    
    display_name = await node_.read_display_name()     
    if org["name"] != display_name.Text:       
        now = datetime.datetime.now()
        await node_.write_attribute(ua.AttributeIds.DisplayName, \
            ua.DataValue(ua.uatypes.LocalizedText(org["name"].replace(':', ' ')), \
            ua.StatusCode(0), now.replace(tzinfo=tz_local).astimezone(tz_utc)))        
        await node_.write_attribute(ua.AttributeIds.BrowseName, \
            ua.DataValue(ua.uatypes.QualifiedName(org["name"].replace(':', ' ')), \
            ua.StatusCode(0), now.replace(tzinfo=tz_local).astimezone(tz_utc))) 
        await node_.write_attribute(ua.AttributeIds.Description, \
            ua.DataValue(ua.uatypes.LocalizedText(org["name"].replace(':', ' ')), \
            ua.StatusCode(0), now.replace(tzinfo=tz_local).astimezone(tz_utc)))`

**Screenshots**<br />
Screenshot attached
[image]

**Version**<br />
Python-Version: 3.11.2 x64<br />
opcua-asyncio Version (e.g. master branch, 0.9): asyncua-1.1.5

AndreasHeine, 2024-12-25:
why do you think its an issue of asyncua? in the attributes tab it shows the current value so from library side its working! 
its more likely an issue in the client software not refreshing! 
maybe close and reopen the parent folder/object so it will rebrowse/reread?

aldoshkind, 2025-01-11:
Hello, @AndreasHeine !
I same exactly same problem as @PetroGolovnya . The node name rendered in tree view is incorrect (old) even after client software restart.
The client software on the picture above is Prosys OPC UA Browser. I use it as well.
It seems that for tree rendering Prosys uses Browse/Display name aquired via ReferenceDescription. Possibly asyncua does not update Browse/Display name stored in node references or does it incorrectly. You can check this by printing refernce in ViewService._browse after node's Browse name has been updated (https://github.com/FreeOpcUa/opcua-asyncio/blob/b1cfae34668044062880930e5caa49cd31ed3a4f/asyncua/server/address_space.py#L149).
Please consider reviewing the issue.

cziebuhr, 2025-02-11:
I had a similar issue. The attributes on the node itself gets updated, but asyncua server stores that information a second time as ReferenceDescription on the parent node. So the result is different if you read data by using node.get_children_descriptions() vs reading by node.read_attribute(). Fazit: changing BrowseName, DisplayName, or TypeDefinition after the node has been created, doesn't work.

Miaou, 2025-05-28:
Hi,

Stumbling on the same issue today. In `common/manage_nodes.py:_create_variable()`:

```python
async def _create_variable(*args, **kwargs):
    addnode = ua.AddNodesItem()
    # [skipped for clarity]
    attrs = ua.VariableAttributes()
    attrs.Description = ua.LocalizedText(qname.Name)
    attrs.DisplayName = ua.LocalizedText(qname.Name)
    # [skipped for clarity]
    addnode.NodeAttributes = attrs
    results = await session.add_nodes([addnode])  # <- calls NodeManagementService
    results[0].StatusCode.check()
    return results[0].AddedNodeId
```

Then `server/address_space.py:NodeManagementService.add_nodes` calls `_add_node` which calls `_add_ref_from_parent`:

```python
    def _add_ref_from_parent(self, nodedata, item, parentdata):
        desc = ua.ReferenceDescription()
        desc.ReferenceTypeId = item.ReferenceTypeId
        desc.NodeId = nodedata.nodeid
        desc.NodeClass = item.NodeClass
        desc.BrowseName = item.BrowseName
        desc.DisplayName = item.NodeAttributes.DisplayName
        desc.TypeDefinition = item.TypeDefinition
        desc.IsForward = True  # FIXME in uaprotocol_auto.py
        self._add_unique_reference(parentdata, desc)  # FIXME return StatusCode is not evaluated
```

This means that, as mentioned by @cziebuhr, there is data duplication and the value for Description and DisplayName is stored both in the address space's nodedata and in the references data.

There seems to be solutions but they are not easy:
1. remove the NodeClass, BrowseName, DisplayName, TypeDefinition from the Reference and fetch them when browsing (it should be possible because we have the target NodeId)
2. when writing attributes, if the attribute is one of these, also update the Reference

Thanks.