FreeOpcUa/opcua-asyncio #1771: Coroutine not awaited while deleting node

aldoshkind, 2025-01-11:
**Describe the bug**
When calling `asyncua_server.delete_nodes([self.node], recursive=True)` there appears message in terminal:
```
.local/lib/python3.12/site-packages/asyncua/server/address_space.py:428: RuntimeWarning: coroutine 'MonitoredItemService.datachange_callback' was never awaited
  callback(handle, None, ua.StatusCode(ua.StatusCodes.BadNodeIdUnknown))
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
```

It seems there are some sync/async issues, maybe some functionality doesn't work properly.

**To Reproduce**
Call `asyncua_server.delete_nodes([self.node], recursive=True)`

**Expected behavior**
No warnings

**Version**
Python-Version: 3.12
opcua-asyncio Version: 1.1.5

oroulet, 2025-01-11:
sounds specific to your code. there not much relation between delete_node command and datachange callbacks, unless you are deleting the nodes you are subscribed to.
I jsut went through thte code and that method is used one place and is awaited...

aldoshkind, 2025-01-12:
Hello, @oroulet ! Thank you for reply!

Method being called is `<bound method MonitoredItemService.datachange_callback of <asyncua.server.monitored_item_service.MonitoredItemService object at 0x7c603ebbfbc0>>`. It is called from `NodeManagementService._delete_node_callbacks` here: https://github.com/FreeOpcUa/opcua-asyncio/blob/b1cfae34668044062880930e5caa49cd31ed3a4f/asyncua/server/address_space.py#L437 and doesn't seem to be awaited. `NodeManagementService._delete_node_callbacks` is sync while `datachange_callback` is async.

`datachange_callback` is added if variable historization is requested:

```
    await self.server.asyncua_server.historize_node_data_change(variable, period=None)
  File ".../.local/lib/python3.12/site-packages/asyncua/server/server.py", line 763, in historize_node_data_change
    await self.iserver.enable_history_data_change(n, period, count)
  File ".../.local/lib/python3.12/site-packages/asyncua/server/internal_server.py", line 295, in enable_history_data_change
    await self.history_manager.historize_data_change(node, period, count)
  File ".../.local/lib/python3.12/site-packages/asyncua/server/history.py", line 269, in historize_data_change
    handler = await self._sub.subscribe_data_change(node)
  File ".../.local/lib/python3.12/site-packages/asyncua/common/subscription.py", line 290, in subscribe_data_change
    return await self._subscribe(
  File ".../.local/lib/python3.12/site-packages/asyncua/common/subscription.py", line 420, in _subscribe
    mids = await self.create_monitored_items(mirs)
  File ".../.local/lib/python3.12/site-packages/asyncua/common/subscription.py", line 476, in create_monitored_items
    results = await self.server.create_monitored_items(params)
  File ".../.local/lib/python3.12/site-packages/asyncua/server/internal_session.py", line 211, in create_monitored_items
    subscription_result = await self.subscription_service.create_monitored_items(params)
  File ".../.local/lib/python3.12/site-packages/asyncua/server/subscription_service.py", line 98, in create_monitored_items
    return await self.subscriptions[params.SubscriptionId].monitored_item_srv.create_monitored_items(params)
  File ".../.local/lib/python3.12/site-packages/asyncua/server/monitored_item_service.py", line 67, in create_monitored_items
    result = await self._create_data_change_monitored_item(item)
  File ".../.local/lib/python3.12/site-packages/asyncua/server/monitored_item_service.py", line 140, in _create_data_change_monitored_item
    result.StatusCode, handle = self.aspace.add_datachange_callback(params.ItemToMonitor.NodeId, params.ItemToMonitor.AttributeId, self.datachange_callback)
```

Here: https://github.com/FreeOpcUa/opcua-asyncio/blob/b1cfae34668044062880930e5caa49cd31ed3a4f/asyncua/server/monitored_item_service.py#L148

closed 2026-11-03 by oroulet