FreeOpcUa/opcua-asyncio #1778: Reading Extensionobjects

sigurdrb, 2025-01-23:
Hi! I have an extensionobject of the type

MyObject Array[32]

[0]:
element1 : BOOL;
element2 : FLOAT;
element3: INT;
[1]
element1 : BOOL;
element2 : FLOAT;
element3: INT;
[2]
element1 : BOOL;
element2 : FLOAT;
element3: INT;
.
.
.
[31]
element1 : BOOL;
element2 : FLOAT;
element3: INT;

I have managed to access the node and stored it as 'MyNode'
however I cant access any of the data by name.
If I want to read element1 of the third item I can do that by
print = MyObject[2].Body[0]
I would rather be able to access it by MyObject[2].element1
or at least be able to read the names of all the elements by a method like read_display_name() or similar.
I have been trying to access the names of the elements in some way or another, but with no luck unfortunately.
If any of you could help, I would greatly appreciate it :)

oroulet, 2025-01-23:
look at the examples. If you call load_type_definition() on server you  should get a python object with aatributes directly accessible. unless things are badly defined on server side

sigurdrb, 2025-01-24:
Thanks for the reply. I used the load_type_definition(). From the documentation it says that "These classes will available in ua module".
Do you know how i can access the values from the ua module?
I tried to follow the example [client_read-custom_structures.py](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client_read-custom_structures.py) but I didnt get any thing different in the before/after example

`[ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' [... cut ...] onObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), ExtensionObject(TypeId=NodeId(Identifier='BilgeObject|defaultbinary', NamespaceIndex=5, NodeIdType=<NodeIdType.String: 3>), Body=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')]`

The structure of the object in this example is an array of 32 structs (called BilgeObject)

[image]

sigurdrb, 2025-01-24:
I can also find the struct from the OPC server here:
I'm sorry if this is a very elementary question. I just cant seem to find a way to connect the values from the object which I can read at 
MyObject[2].Body[0]  (MyObject is an array of 32 objects that is called BilgeObject.
Was hoping to reference them to names as available, enable, etc (as picture below).
[image]

[image]

schroeder-, 2025-01-24:
Try using `load_data_type_definitions` instead of `load_type_definition`.

sigurdrb, 2025-01-24:
Ok, I see. But I should get it to work with the old method? I got it to work with the examples in github, but not on the server I work with. :( 
But UaExpert managed to read the structure, so I guess I should get it to work with the old method?

closed 2025-01-28 by sigurdrb