FreeOpcUa/opcua-asyncio #1783: Can't write values... datatype error?

GitHub01Aetech, 2025-01-31:
Hello everyone,

I'm trying to write a variable in a tag/node of the PLC with asyncio and I can't do it... I don't know what could be happening, if you know something that could help me I would appreciate it.

I've tried both with write_values([nodeIds], [valuesToWrite]) and with write_value(nodeId, valueToWrite) and it's not working...

Here is the part of the code, let me explain you:

# results = await globals.opc_client.write_values(nodes_to_write, nodesValues_to_write)
        
    for node, value in zip(nodes_to_write, nodesValues_to_write):
        print("node in zip", node)
        print("value in zip", value)
        result = await node.write_value(value)
        # for result in results:
        if result.StatusCode.is_good():
            print("Write successful")
        else:
            print("Error writing tag dentro de result:", result.StatusCode)

**node in zip:** 
[ns=3;s="EQ11_PH1"."TO"."TEST1"]

**value in zip:** 
DataValue(Value=Variant(Value='1', VariantType=<VariantType.Int16: 4>, Dimensions=None, is_array=False), StatusCode_=StatusCode(value=0), SourceTimestamp=None, ServerTimestamp=None, SourcePicoseconds=None, ServerPicoseconds=None)

**And I get that error:**

Error writing tag: required argument is not an integer

**I try to write into the node directly like this too:** 

        result = await node.write_value(85)

**And then I get that error:** 
"Error writing tag: The server does not support writing the combination of value, status and timestamps provided.(BadWriteNotSupported)"

It seems something happens in the dataValue, that's my code to get it:

valorOK2 = ua.DataValue(ua.Variant(value, ua.VariantType(switch_opc_data_type(tag['DATATYPE']))))

where datatype is FLOAT,INT,DEC, etc... and I switch to a valid opc_data_type:

from asyncua  import ua

def switch_opc_data_type(datatype):
    datatype = datatype.lower()
    if datatype == "int":
        return ua.VariantType.Int16.value
    elif datatype == "dint":
        return ua.VariantType.Int32.value
    elif datatype == "str":
        return ua.VariantType.String.value
    elif datatype == "dec":
        return ua.VariantType.Float.value
    elif datatype == "ddec":
        return ua.VariantType.Double.value
    elif datatype == "dat":
        return ua.VariantType.DateTime.value
    else:
        return ua.VariantType.String.value

That's all I think... so if someone can help me I would appreciate it! 

Thanks!

AndreasHeine, 2025-01-31:
duplicate!

https://github.com/FreeOpcUa/opcua-asyncio/issues?q=The%20server%20does%20not%20support%20writing%20the%20combination%20of%20value%2C%20status%20and%20timestamps%20provided.

GitHub01Aetech, 2025-01-31:
I don't think is a duplicate one because I already do that things from the other issues and I can't reach the write:

I already import ua from asyncua

`from asyncua import Client, ua`

I can't do this in my ua.DataValue, for example:

```

dv = ua.DataValue(ua.Variant(1.2, ua.VariantType.Double))
                dv.ServerTimestamp = datetime.now()
                dv.SourceTimestamp = datetime.now()
```

And I get this error: 
err cannot assign to field 'ServerTimestamp'

do you have some other advise? it's seems all looks good... but I can't get the error

AndreasHeine, 2025-01-31:
looks like a siemens plc e.g. S7 1200/1500

there are only two options:

timestamps = None
```python
        dv = ua.DataValue(
            Value=ua.Variant(100.0, ua.VariantType.Double),
            StatusCode_=ua.StatusCodes.Good,
            ServerTimestamp=None,
            SourceTimestamp=None
        )
```

timestamps set!
```python
        dv = ua.DataValue(
            Value=ua.Variant(100.0, ua.VariantType.Double),
            StatusCode_=ua.StatusCodes.Good,
            ServerTimestamp=datetime.datetime.now(datetime.timezone.utc),
            SourceTimestamp=datetime.datetime.now(datetime.timezone.utc)
        )
        print("W
```

otherwise time/zone is something else...

PLC always have issues with time so most don't like timestamps...
If you don't know the VariantType and have only the NodeId and a Value stored you can `node.read_data_value()` before writing in order to know the correct datatype!!!