FreeOpcUa/opcua-asyncio #1787: Speeding up OPC-UA node traversal from hours to minutes

kzawad1-ces, 2025-02-12:
**Describe the bug** <br />
I am using "asyncua" library to connect to an OPC-UA server and get list of all of the tags available on that server.  The Python application takes almost 4 hours to traverse the OPC-UA server, here is the output of the app:

```
Starting OPC-UA Client Application

Started at: 2025-02-09 06:28:42.764038 
URL: opc.tcp://172.16.1.5:62541
File (Output): Tags_2025-02-09_062842_UTC.txt

Client(url=url)
Set User
Set Password
Set Application_URI
Client set_security
USE_TRUST_STORE: True
if USE_TRUST_STORE
async with client -> get nodes
Called client.node.root
Called to get_child 0:Objects
Call asyncio.wait_for()

Runtime Information
Start Time: 2025-02-09 06:28:42.764038 
End Time:   2025-02-09 10:22:53.466855 
Duration:   3:54:10.702817
```

The code calls:

`await extract_node_paths(objects, "/Objects", output_file)`

and the main function looks like this:

```
async def extract_node_paths(node, path, output_file):
    children = await node.get_children()
    for child in children:
        browse_name = await child.read_browse_name()
        child_path = f"{path}/{browse_name.Name}"
        if await child.read_node_class() == 2:  # Variable node class
            with open(output_file, 'a') as f:
                f.write(child_path + '\n')
        await extract_node_paths(child, child_path, output_file)
```

The output has the following syntax:

```
/Objects/Tag Providers/ZLT/ZLT/Dispatcher/Configuration/Limits/LimitBESS
/Objects/Tag Providers/ZLT/ZLT/Dispatcher/Configuration/Limits/LimitEnergyBESS
/Objects/Tag Providers/ZLT/ZLT/Dispatcher/Configuration/Limits/LimitPOI
```

The output file has 104,529 lines.  So 104,529 tags.  

The question I have is:

1. Is it possible to speed this up where it can do this work in 20 minutes? 
2. Is there a built-in API can get a tree list of the tags is much less time?

**To Reproduce**<br />
Code is shown above.

**Expected behavior**<br />
I was expecting that there would be a relatively fast (minutes and not hours) way to get a OPC-UA server tags list.

**Screenshots**<br />
None, but can get some if needed.

**Version**<br />
Python-Version: Python 3.10.12<br />
opcua-asyncio Version (e.g. master branch, 0.9):  asyncua==1.1.5

AndreasHeine, 2025-02-13:
i would use:

here is somthing similar... but its for the subscription to chunck if the operational limits of the server are smaller then the amount of nodes i want to subscribe to... but you can build it for a read/browse call as well
```python
async def subscribe(
    subscription: Subscription, 
    nodes: Union[Node, Iterable[Node]], 
    queuesize = 0,
    maxpercall: int = None, 
    attr = ua.AttributeIds.Value,
    monitoring = ua.MonitoringMode.Reporting
):
    '''
    Subscribing to large numbers of nodes without overwhelming the server!
    '''
    if not isinstance(nodes, list):
        nodes = [nodes]

    if not maxpercall:
        try:
            # try to read the "MaxMonitoredItemsPerCall" value
            maxpercall = await subscription.server.get_node("i=11714").read_value()
            if maxpercall == 0: 
                maxpercall = None
        except:
            maxpercall = None

    if not maxpercall:
        return await subscription.subscribe_data_change(nodes)
    else:
        handle_list = []
        for each in [nodes[i:i + maxpercall] for i in range(0, len(nodes), maxpercall)]:
            handles = await subscription.subscribe_data_change(
                nodes=each,
                attr=attr,
                queuesize=queuesize,
                monitoring=monitoring,
            )
            handle_list.append(handles)
        return handle_list
```

oroulet, 2025-02-13:
Arrrg when I got that email I hoped it was a MR ;-).

oroulet, 2025-02-13:
And yes do not use use several network calls per node. Use Node.get_children_desc() or whatever it is called. That will reduce time. 
And then doing these things is slow anyway due to network calls

kzawad1-ces, 2025-02-13:
@AndreasHeine and @oroulet , I get all of your criticisms.  

Yes, there is a lot to consider here.  I've written hundreds of thousands of lines of code in C/C++.  I get it about writing to a file every time and network traffic for parsing the nodes, but my approach with this is not from a purely programming mindset.  The reason I am writing to a file every time is because if the app runs for 2 hours and gets an exception, I lost all the tags. 

It is with an end outcome mindset.  I look at tools like Unified Automation Expert and others and they don't have a way to dump a tag list to file and I thought, someone must have figure this out already.  

I will give you subscribe function a try.  The 20+ OPC-UA servers I am working with are configured with Ignition SCADA, and the server are configured in so a way that the client has no way to request polling or subscription at certain rates.  The server configures the policy and pushes it down.  The servers have over 100k tags.

Now, that I have a baseline.  I want to see how your function compares to the baseline.

AndreasHeine, 2025-02-13:
fix for sync file stuff:
just use aiofile instead of sync stuff... mixing sync and asyncio code is always a bad idea (its either or!!!)
see: https://pypi.org/project/aiofile/

fix for unnessesary calls (browse):
```python
#    for child in children:
#        browse_name = await child.read_browse_name() 
bname_data_values = await client.read_attributes(children, ua.AttributeIds.BrowseName)

# if await child.read_node_class() == 2:
nodeclass_data_values = await client.read_attributes(children, ua.AttributeIds.NodeClass)
# do your for loop stuff here
```

remember read_attributes returns DataValue classes!!!
https://github.com/FreeOpcUa/opcua-asyncio/blob/772316de21df3b0f874c144b873950f96e188c03/asyncua/client/client.py#L929

kzawad1-ces, 2025-02-13:
For this particular Ignition SCADA OPC-UA server, the values for 

MaxNodesPerBrowse: 250
MaxNodesPerRead: 10000

[image]

AndreasHeine, 2025-02-13:
should look like this:
```python
        # maxpercall = the value of "MaxNodesPerBrowse" your client can read that after connection!
        results = []
        for each in [nodes[i:i + maxpercall] for i in range(0, len(nodes), maxpercall)]:
            res = await await client.read_attributes(children, ua.AttributeIds.BrowseName)
            results .append(res )
```
and 
```python
        # maxpercall = the value of "MaxNodesPerRead" your client can read that after connection!
        results = []
        for each in [nodes[i:i + maxpercall] for i in range(0, len(nodes), maxpercall)]:
            res = await await client.read_attributes(children, ua.AttributeIds.NodeClass)
            results .append(res )
```

milosmatovic, 2025-02-27:
Out of curiosity, is there any reason for executing `read_attributes` one by one? If we want async then it should be something like this, or am i missing something? 
```python
# maxperbrowse = the value of "MaxNodesPerBrowse" your client can read that after connection!
tasks= []
for each in [children[i:i + maxperbrowse] for i in range(0, len(children), maxperbrowse)]:
    tasks.append(client.read_attributes(each, ua.AttributeIds.BrowseName))

results = asyncio.gather(*tasks, return_exceptions=True)

for res in results:
    if isinstance(res, BaseException):
        # Handle error here
    else:
        # Process result here
```

kzawad1-ces, 2025-03-04:
@AndreasHeine is there a way to change the settings so that two items happen:

- If connection is interrupted it reconnects and resumes.
- Also, which object (loop or client object) would I need to change the timeout, so I don't get the TimeoutError?

```
Called to get_child 0:Objects
Call asyncio.wait_for()
Traceback (most recent call last):
  File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File ".../src/ces_opcua_async_client_v2_ug_ar.py", line 361, in <module>
    ces_opcua_client__main()
  File ".../src/ces_opcua_async_client_v2_ug_ar.py", line 351, in ces_opcua_client__main
    loop.run_until_complete(task(loop, file_name))
  File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File ".../src/ces_opcua_async_client_v2_ug_ar.py", line 183, in task
    await extract_node_paths(objects, "/Objects", output_file)
  File ".../src/ces_opcua_async_client_v2_ug_ar.py", line 58, in extract_node_paths
    await extract_node_paths(child, child_path, output_file)
  File ".../src/ces_opcua_async_client_v2_ug_ar.py", line 58, in extract_node_paths
    await extract_node_paths(child, child_path, output_file)
  File ".../src/ces_opcua_async_client_v2_ug_ar.py", line 58, in extract_node_paths
    await extract_node_paths(child, child_path, output_file)
  File ".../src/ces_opcua_async_client_v2_ug_ar.py", line 51, in extract_node_paths
    children = await node.get_children()
               ^^^^^^^^^^^^^^^^^^^^^^^^^
  File ".../src/venv/lib/python3.12/site-packages/asyncua/common/node.py", line 390, in get_children
    return await self.get_referenced_nodes(refs, ua.BrowseDirection.Forward, nodeclassmask)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File ".../src/venv/lib/python3.12/site-packages/asyncua/common/node.py", line 487, in get_referenced_nodes
    references = await self.get_references(refs, direction, nodeclassmask, includesubtypes)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File ".../src/venv/lib/python3.12/site-packages/asyncua/common/node.py", line 457, in get_references
    results = await self.session.browse(params)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File ".../src/venv/lib/python3.12/site-packages/asyncua/client/ua_client.py", line 384, in browse
    data = await self.protocol.send_request(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File ".../src/venv/lib/python3.12/site-packages/asyncua/client/ua_client.py", line 167, in send_request
    data = await asyncio.wait_for(self._send_request(request, timeout, message_type), timeout if timeout else None)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
```

oroulet, 2025-03-13:
Client constructor timeout.
automatic re-conneciton is not a feature

kzawad1-ces, 2025-03-17:
The timeout exception is coming from the "run_intil_complete()", the source code is the following:

```
# ------------------------------------------------------------------------------
def exec_loop(opcua_obj, file_name):
    '''
    Asyncio Loop wrapper function.
    '''

    loop = asyncio.new_event_loop()
    asyncio.set_event_loop(loop)

    try:
        loop.set_debug(True)

        loop.run_until_complete(task(opcua_obj, file_name))
        #loop.run_until_complete(task(opcua_obj, file_name), timeout=43200)
        #loop.wait_for(task(opcua_obj, file_name), timeout=43200)

        
    except TimeoutError as exp:
        print('Exception Start ------------')
        print('TimeoutError')
        print(str(exp))
        print('Exception End ------------')
    except KeyboardInterrupt as exp:
        print('Exception Start KeyboardInterrupt ------------')
        print('KeyboardInterrupt')
        print(str(exp))
        print('Exception End KeyboardInterrupt ------------')
        pass
    finally:
        loop.close()
```

The output shows the debug statements I place there, as shown below:
```
Exception Start ua.UaError x ------------
exp:The node id refers to a node that does not exist in the server address space.(BadNodeIdUnknown)
path:/Root/Objects/Tracker50A/RTU/_Hints
Exception End ua.UaError x ------------
Exception Start ua.UaError x ------------
exp:The node id refers to a node that does not exist in the server address space.(BadNodeIdUnknown)
path:/Root/Objects/Tracker51A/RTU/_Hints
Exception End ua.UaError x ------------
Exception Start ua.UaError x ------------
exp:The node id refers to a node that does not exist in the server address space.(BadNodeIdUnknown)
path:/Root/Objects/Tracker51B/RTU/_Hints
Exception End ua.UaError x ------------
Exception Start ------------
TimeoutError

Exception End ------------

Runtime Information
Start Time: 2025-03-15 06:03:40.399472 
End Time:   2025-03-16 03:40:20.975995 
Duration:   21:36:40.576523

```

It is not coming from "client" as you say because I added different debug prints there:

```
    try:
        print('async with client -> get nodes')
        async with client:
            print('Called client.node.root')
            root = client.nodes.root
            print('Called to get_child 0:Root')
            await extract_node_paths(objects, "/Root", output_file)
            #await asyncio.wait_for(extract_node_paths(objects, "/Objects", output_file), timeout)

    except ua.UaError as exp:
        print('Exception Start a.UaError ------------')
        _logger.error(exp)
        print('Exception End a.UaError ------------')

```

So the question is why the exception is being caught at the asyncio top level, it should be running until completion.  Otherwise the error would be outputting "Exception Start a.UaError".