FreeOpcUa/opcua-asyncio #1793: Traeger as the OPC Server

mikependon, 2025-03-06:
I am trying to understand how OPC compatibility works, but maybe you can help us with this.

In our organization, we have a simulator that uses the [Traeger](https://www.traeger.de/en/products/development/opcua/opcua-sdk) as the OPC UA Server. When we try to connect to it using this library, it is throwing a `TimeoutError`. The error happens on the `open_secure_channel()` method of the `us_client.py` file.

Traeger OPC Server: https://www.traeger.de/en/products/development/opcua/opcua-sdk

As I understand it, for as long the OPC Server is within the same UA protocol, this library should work, irregardless to which language the OPC Server is written. Does this library supports the Traeger as the OPC UA server?

[image]

Can you help us understand or verify whether this library can support them? Thanks a lot.

AndreasHeine, 2025-03-06:
@mikependon 

does it work with UAExpert or Prosys OPC UA Browser?

next is to activate the logger to see what happens before the CancelledError:

```python
logging.basicConfig(level=logging.DEBUG)
_logger = logging.getLogger('asyncua')
```

mikependon, 2025-03-06:
Yes, we have Prosys OPC UA Browser installed and it works there. There is no credentials on the OPC server, so it is just a direct and simple connection for now.
Note: The same code we have are working if we use the freeopcua `Server`.

mikependon, 2025-03-06:
Thanks for the logging trick, I do not know that. Here is the log.

```
DEBUG:asyncio:Using proactor: IocpProactor
INFO:asyncua.client.client:connect
INFO:asyncua.client.ua_client.UaClient:opening connection
INFO:asyncua.uaprotocol:updating client limits to: TransportLimits(max_recv_buffer=65535, max_send_buffer=65535, max_chunk_count=0, max_message_size=10485760)
INFO:asyncua.client.ua_client.UASocketProtocol:open_secure_channel
DEBUG:asyncua.client.ua_client.UASocketProtocol:Sending: OpenSecureChannelRequest(TypeId=FourByteNodeId(Identifier=446, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>), RequestHeader_=RequestHeader(AuthenticationToken=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>), Timestamp=datetime.datetime(2025, 3, 6, 11, 59, 37, 760566, tzinfo=datetime.timezone.utc), RequestHandle=1, ReturnDiagnostics=0, AuditEntryId=None, TimeoutHint=1000, AdditionalHeader=ExtensionObject(TypeId=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>), Body=None)), Parameters=OpenSecureChannelParameters(ClientProtocolVersion=0, RequestType=<SecurityTokenRequestType.Issue: 0>, SecurityMode=<MessageSecurityMode.None_: 1>, ClientNonce=b'', RequestedLifetime=3600000))
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
Error: TimeoutError:
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection
```

mikependon, 2025-03-06:
OK, thanks. Let me check with having the security.

oroulet, 2025-03-06:
are you sure the problem is not simply that the server is slow to answer?
I remember having the exact same issue against one and simply increasing the timeout in Client constructor solved it

mikependon, 2025-03-06:
@oroulet  we have the Traeger OPC UA server up and running. I can even `tnc` both the ip-address and port.

oroulet, 2025-03-06:
Did I say they were not running? Try to increase that timeout
 That is easy

closed 2025-03-12 by mikependon