FreeOpcUa/opcua-asyncio #1795: How to access nodes without using get_child method?

danmoser, 2025-03-10:
With prior versions of asyncua (<1.0.x) I was able to access nodes with as ``var = client.get_node(ua.NodeId(1002, 2))`` or ``var = client.get_node("ns=3;i=2002")``. How these always return ``BadNodeIdUnknown``!

My server has many layers of nodes and using ``get_child()`` is not convenient...
TIA

AndreasHeine, 2025-03-10:
@danmoser 
get_node and get_child are completly different things...
get_node does nothing on the wire (creates just a Node-Class instance without any request)
get_child makes a browserequest for hierarical references

BadNodeIdUnknown -> it is likely that the NamespaceIndex has changed for your node...

https://www.linkedin.com/pulse/how-avoid-badnodeidunknown-known-working-tags-andreas-heine-kvege/?trackingId=VH473M%2Byw%2F9B9LqWeVHA9g%3D%3D

danmoser, 2025-03-10:
@AndreasHeine: thanks for quick response... But the issue is:

My code using "get_node" works w/ asyncua  (<1.0.x) and stopped working w/ asyncua 1.1.x. The newer version returns ther "BadNodeIdUnknown" error... Can the newer version of asyncua change the NamespaceIndex?

danmoser, 2025-03-10:
I found the issue... It was a tailing linebreak (``\n``) which asyncua 1.0.x was ignoring it.
```
myname = "mynodeset\n"
mystr = f"ns=2;s={myname}"
ns = client.get_node(mystr)
print("Node name: ", await ns.read_browse_name())
```
Thank you!