FreeOpcUa/opcua-asyncio #1823: Connect to prosys opc ua simulation server failed

unregistered33, 2025-04-19:
Hi! Im try to connect to prosys opc ua server with example:
[(https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client_to_prosys.py)]
but have error:
"
................
  File "C:\Users\unreg\AppData\Local\Programs\Python\Python311\Lib\enum.py", line 695, in __call__
    return cls.__new__(cls, value)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\unreg\AppData\Local\Programs\Python\Python311\Lib\enum.py", line 1111, in __new__
    raise ve_exc
ValueError: 36 is not a valid NodeIdType"

Prosys Server version 5.5.2-362
SDK version 5.2.8-159

This is very strange, because about six months ago the connection was fine (to the old version of the server).
The client UAexpert 1.6.3 448 work fine
How i can fix it? Thanks!

oroulet, 2025-04-25:
that was not easy but I finally found out the new prosys server now requires a certificate even without any encryption... and we did not support that...
fixed in https://github.com/FreeOpcUa/opcua-asyncio/pull/1827

MarkSSH, 2025-05-12:
Hi, I'm also try to connect to prosys opc ua server with the fixed example, also got an error:
ValueError: 36 is not a valid NodeIdType

Is this error really be fixed?

oroulet, 2025-05-12:
supposed to be fixed in master yes

MarkSSH, 2025-05-12:
if i dev in my own computer, and also run a prosys opc ua server, should i generation an own "my_cert.der" file?
Can i just use "my_cert.der" file in master without change anything?

MarkSSH, 2025-05-12:
when i downgrade asyncua to 0.9.96, it can be resolved!

but i want to know how it work?

pal-software, 2025-07-13:
I have same problem...

In short, problem at the field `security_policy.host_certificate` of class `Client` .
When host certificate not defined (None), then now its initialized by `b""`.
But in prior versions its initialized by `None`.

For me.. Now I make quick changes at the file `ayncua/crypto/uacrypto.py` at method `der_from_x509`, from this
```python
def der_from_x509(certificate):
    if certificate is None:
        return b""
    return certificate.public_bytes(serialization.Encoding.DER)
```
to this
```python
def der_from_x509(certificate):
    if certificate is None:
        return None
    return certificate.public_bytes(serialization.Encoding.DER)
```
this method called by `class SecurityPolicyNone(SecurityPolicy):`  (method `__init__`) for initialization its properties.

but... mybe its wrong & more right is change not this method directly

sorry my poor english :)