FreeOpcUa/opcua-asyncio #1824: Create Nodes During Subscription

benmusson, 2025-04-23:
## Problem Statement
I need to model a large namespace that will not support browsing. 

I would like to create my nodes for only as long as at least one subscription exists.  

## What I've tried

I've tried subscribing to the `ItemSubscriptionCreated` callback, and to use the callback as a chance to create my nodes.
```python
# during setup
self.server.subscribe_server_callback(
    CallbackType.ItemSubscriptionCreated, self._dispatch_item_subscription
)

...

# _dispatch_item_subscription
# Create a new node, and set the status of the `ItemSubscriptionCreated` response_params.
event.response_params[index].StatusCode = (
    await component.on_item_subscription(item)
)
``` 

This almost works. 

The Good: The nodes are created and the client receives whatever status code I return. 
The Bad: The client's subscription does not seem to work (i.e. receive values). Only after creating a second subscription does it work as expected.

I believe this is because although I'm setting the `StatusCode`, my callback is being run after the subscription creation handling logic, meaning the Node doesn't exist yet to properly create a subscription.

## Questions
1. Does this approach make any sense?
2. Is there another way to accomplish a dynamic address space? 

I believe Milo supports this in the form of a [ManagedAddressSpace](https://javadoc.io/doc/org.eclipse.milo/sdk-server/0.3.3/org/eclipse/milo/opcua/sdk/server/api/ManagedAddressSpace.html) that gives you exact control over the handling of service calls.

oroulet, 2025-04-25:
IS that part of ua protocol?
Does it need changed in the asyncua code?
you can always make a PR to show how such a thing can be done but I am unsure if it should be merged.

benmusson, 2025-04-25:
In my understanding, the standards do not care how a namespace is practically implemented. There's nothing that says nodes need to live in memory or that they have to "exist" before they are read/subscribed. 

There are advantages to having a dynamic/virtual namespace, but this library is not really structured around that concept. 

I believe this could be accomplished if we had a route to override or reimplement methods of the AddressSpace class.

closed 2025-04-25 by benmusson