FreeOpcUa/opcua-asyncio #1842: Missing optional session_timeout during connecting in context manager

rkrell, 2025-06-11:
**Describe the bug** <br />
On several devices I get this warning:
_Requested session timeout to be 3600000ms, got 30000ms instead_
This happens while connecting this way:
```python
async with Client(url=opc_url) as client:
  ...
```
Setting
`client.session_timeout = 30000`
in this context has no effect, of course.

**To Reproduce**<br />
Steps to reproduce the behavior incl code:

Connecting this way to a Siemens S7-1500 device:

```python
from asyncua import Client

async with Client(url=opc_url) as client:
    ...
```

**Expected behavior**<br />
The connection should establish without a warning.

For achieving this, explicitly setting and optional parameter would help:
```python
from asyncua import Client

async with Client(url=opc_url, session_timeout=30000) as client:
    ...
```
but this is not supported, the session_timeout parameter is statically set to:
```python
self.session_timeout = 3600000  # 1 hour
```
**Version**<br />
Python-Version: 3.11<br />
opcua-asyncio Version: 1.1.6:

schroeder-, 2025-06-11:
Directly using the Client only works for basic use cases. 
Better init the object and then use it with async with.

```
from asyncua import Client
...

client = Client(url=opc_url)
client.session_timeout = 30000
async with client:
          .....
```

rkrell, 2025-06-11:
Thank you for explanation