FreeOpcUa/opcua-asyncio #1844: Cannot connect for server without security

lalitMCrion, 2025-06-18:
`async def task(loop):
    url = "opc.tcp://myserver:4840"

    try:
        client = Client(url=url)
        client.set_user("User")
        client.set_password("Test")

        await client.connect()
        print("connected to OPC UA Server")
    except Exception:
        _logger.exception("error")
    finally:
        await client.disconnect()
`
URI and Creds are Dummy, Iam just using the standard [https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client-minimal-auth.py](auth with no security example). I am able connect to server using "UA Expert" also using the .net opc ua. 

Getting following error

> INFO:asyncua.client.client:connect
INFO:asyncua.client.ua_client.UaClient:opening connection
INFO:asyncua.uaprotocol:updating client limits to: TransportLimits(max_recv_buffer=65535, max_send_buffer=65535, max_chunk_count=0, max_message_size=0)
INFO:asyncua.client.ua_client.UASocketProtocol:open_secure_channel
INFO:asyncua.client.ua_client.UaClient:create_session
INFO:asyncua.client.ua_client.UASocketProtocol:close_secure_channel
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
ERROR:asyncua:error
Traceback (most recent call last):
  File "/Users/lalitm/Work/Python/opcaua/client_connect.py", line 17, in task
    await client.connect()
  File "/Users/lalitm/Work/Python/opcaua/.venv/lib/python3.12/site-packages/asyncua/client/client.py", line 321, in connect
    await self.create_session()
  File "/Users/lalitm/Work/Python/opcaua/.venv/lib/python3.12/site-packages/asyncua/client/client.py", line 510, in create_session
    response = await self.uaclient.create_session(params)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lalitm/Work/Python/opcaua/.venv/lib/python3.12/site-packages/asyncua/client/ua_client.py", line 367, in create_session
    response.ResponseHeader.ServiceResult.check()
  File "/Users/lalitm/Work/Python/opcaua/.venv/lib/python3.12/site-packages/asyncua/ua/uatypes.py", line 383, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadCertificateInvalid: The certificate provided as a parameter is not valid.(BadCertificateInvalid)
INFO:asyncua.client.client:disconnect
INFO:asyncua.client.ua_client.UaClient:close_session
WARNING:asyncua.client.ua_client.UaClient:close_session but connection wasn't established
WARNING:asyncua.client.ua_client.UaClient:close_secure_channel was called but connection is closed
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection

schroeder-, 2025-06-18:
The error says you need to provide a client certificate. Add this to the client setup.

```
import socket
from pathlib import Path
from cryptography.x509.oid import ExtendedKeyUsageOID
from asyncua.crypto.security_policies import SecurityPolicyNone
from asyncua.crypto.cert_gen import setup_self_signed_certificate

....
    cert = Path("certificate.der")
    private_key = Path("key-example.pem")
    host_name = socket.gethostname()
    client_app_uri = f"urn:{host_name}:foobar:myselfsignedclient"

    await setup_self_signed_certificate(
        private_key,
        cert,
        client_app_uri,
        host_name,
        [ExtendedKeyUsageOID.CLIENT_AUTH],
        {
            "countryName": "CN",
            "stateOrProvinceName": "AState",
            "localityName": "Foo",
            "organizationName": "Bar Ltd",
        },
    )
    client = Client(url=url)
    client.application_uri = client_app_uri
    await client.set_security(
        SecurityPolicyNone,
        certificate=str(cert),
        private_key=str(private_key)
    )
    ....
```

closed 2025-06-18 by schroeder-

lalitMCrion, 2025-06-18:
But my server expects no certificates, even then i should create a client certificate and send it?

lalitMCrion, 2025-06-18:
I would like to get little more clarity. I am able to connect to same server using the below code, where I am not creating any certs. But why do I have to do in python. 
`async Task InitializeSession()
    {
        var config = new ApplicationConfiguration()
        {
            ApplicationName = "UnityUAClient",
            ApplicationType = ApplicationType.Client,
            ApplicationUri = $"urn:{System.Net.Dns.GetHostName()}:UnityUAClient",
            SecurityConfiguration = new SecurityConfiguration
            {
                ApplicationCertificate = new CertificateIdentifier(),
                AutoAcceptUntrustedCertificates = true,
                RejectSHA1SignedCertificates = false,

                // Set dummy store paths to avoid exception
                TrustedIssuerCertificates = new CertificateTrustList
                {
                    StoreType = "Directory",
                    StorePath = "CertificateStores/UA Certificate Authorities"
                },
                TrustedPeerCertificates = new CertificateTrustList
                {
                    StoreType = "Directory",
                    StorePath = "CertificateStores/UA Applications"
                },
                RejectedCertificateStore = new CertificateTrustList
                {
                    StoreType = "Directory",
                    StorePath = "CertificateStores/RejectedCertificates"
                }
            },
            TransportConfigurations = new TransportConfigurationCollection(),
            TransportQuotas = new TransportQuotas { OperationTimeout = 15000 },
            ClientConfiguration = new ClientConfiguration { DefaultSessionTimeout = 60000 },
            TraceConfiguration = new TraceConfiguration()
        };

        // Bypass certificate check
        config.CertificateValidator = new CertificateValidator();
        config.CertificateValidator.CertificateValidation += (s, e) =>
        {
            e.Accept = true;
        };

        await config.Validate(ApplicationType.Client);

        var endpoint = CoreClientUtils.SelectEndpoint(serverUrl, useSecurity: false);
        var endpointConfig = EndpointConfiguration.Create(config);
        var endpointDescription = new ConfiguredEndpoint(null, endpoint, endpointConfig);

        var userIdentity = new UserIdentity(username, password);
        Session = await Session.Create(config, endpointDescription, false, "UnitySession", 60000, userIdentity, null);
        Debug.Log($"[UA] Connected: {Session.SessionName}, Security: {Session.Endpoint.SecurityPolicyUri}");
        IsConnected = true;
    }`

lalitMCrion, 2025-06-19:
To give you some context. I'm trying to connect OPC UA to Unity for that I used c# version [.net standard opc ua](https://www.nuget.org/packages/OPCFoundation.NetStandard.Opc.Ua/). But Im facing few issues with regards to Unity Builds. Thats the reason my trying to create a bridge between Unity to OPC UA using the Python, where python connects to OPC UA and sends data to Unity using webscoket.