FreeOpcUa/opcua-asyncio #1852: Unable to find the right connection settings

ts4iot, 2025-07-23:
I try to login to an OPC-UA server with the following connection parameter (see attached picture with UAExpert settings), but I find no way to configure the asyncua client that it works.

The connection with UAExpert works only with this connection settings. No other combination is possible.

[image]

Has anyone expirience with this setting combination and can help me to configure asyncua client in the right way.

Thanks

AndreasHeine, 2025-07-23:
maybe post your simplified client code?

ts4iot, 2025-07-24:
this is one of my attempts:
``````
import asyncio
from asyncua import Client, ua
from asyncua.crypto.security_policies import SecurityPolicyBasic256Sha256

async def task(loop):
    url = "opc.tcp://192.168.0.3:4862"
    client = Client(url=url)
    await client.set_security(
        SecurityPolicyBasic256Sha256,
        mode=ua.MessageSecurityMode.SignAndEncrypt,
        certificate="",
        private_key="",
        server_certificate="dml-002_cert.der")
    await client.connect()
    async with client:
        root = await client.get_root_node()
        print(root)
        await asyncio.sleep(1)
   
def main():
	loop = asyncio.get_event_loop()
	loop.set_debug(True)
	loop.run_until_complete(task(loop))
	loop.close()

if __name__ == "__main__":
	main()
``````

The error in this example is that the set_security expects certificate and key, but I don't have one. How can I combine SecurityPolicyBasic256Sha256, SignAndEncrypt and Anonymous?