FreeOpcUa/opcua-asyncio #1864: `await self._session_watchdog_task` raises RuntimeError

dakotahorstman, 2025-09-05:
**Describe the bug** <br />
I'm not sure when `UaProcessor.close` is called or under what conditions, but `await self._session_watchdog_task` raises the following error:

```python
[2025-09-04 16:40:27,385] [1] [WARNING] Session timed out after 25.0s of inactivity 
[2025-09-04 16:40:36,224] [1] [ERROR] Unexpected crash in BinaryServer._close_tasks 
Traceback (most recent call last):
  File "/opt/pysetup/.venv/lib/python3.12/site-packages/asyncua/server/binary_server_asyncio.py", line 182, in _close_tasks
    await task
  File "/opt/pysetup/.venv/lib/python3.12/site-packages/asyncua/server/uaprocessor.py", line 532, in close
    await self._session_watchdog_task
  File "/opt/pysetup/.venv/lib/python3.12/site-packages/asyncua/server/uaprocessor.py", line 547, in _session_watchdog_loop
    await self.close()
  File "/opt/pysetup/.venv/lib/python3.12/site-packages/asyncua/server/uaprocessor.py", line 532, in close
    await self._session_watchdog_task
RuntimeError: Task cannot await on itself: <Task pending name='Task-129415' coro=<UaProcessor._session_watchdog_loop() running at /opt/pysetup/.venv/lib/python3.12/site-packages/asyncua/server/uaprocessor.py:547>>
```

Whether the follow observation is downstream of this, I'm not sure, but in my application, I've implemented an in-place update mechanism for my tags. It more or less drops and immediately replaces the targeted tag, preserving the node id. One of these operations ran ~15mins prior to this error, but that tag, after replacement, no longer triggered things like data change events. It is the only tag that more-or-less "stopped working".

The tag could be read and written to normally by my PLC, but my application never received data change events so the data was never recorded.

This error is somewhat rare, so some condition must exist for it to occur. Restarting the OPC server fixes the issue whenever it does occur, so I can continue doing that for now.

Let me know if there are any other details needed. More than willing to co-debug this if needed :)

**Version**<br />
Python-Version:<br />3.12.11
opcua-asyncio Version (e.g. master branch, 0.9): 1.1.6

Jellekon, 2026-09-14:
I have similar behaviour

May 27 02:52:21 m-vm-jellelinux python[6679]: 02:52:21 [WARNING] asyncua.server.uaprocessor: Session timed out after 3600.0s of inactivity
May 27 03:20:40 m-vm-jellelinux python[6679]: 03:20:40 [ERROR] asyncua.server.binary_server_asyncio: Unexpected crash in BinaryServer._close_tasks
May 27 03:20:40 m-vm-jellelinux python[6679]:   File "/home/prdmz/projects/OPCUAsimulator/factory-sim/venv/lib/python3.12/site-packages/asyncua/server/binary_server_asyncio.py", line 182, in _close_tasks
May 27 03:20:40 m-vm-jellelinux python[6679]: RuntimeError: Task cannot await on itself: <Task pending name='Task-38066' coro=<UaProcessor._session_watchdog_loop() running at /home/prdmz/projects/OPCUAsimulator/factory-sim/venv/lib/python3.12/site-packages/asyncua/server/uaprocessor.py:548>>

Resulting in memory adding up until application crashes

Error occurs right after connection loss and reconnect

Name: asyncua
Version: 1.1.8
Summary: Pure Python OPC-UA client and server library
Home-page: http://freeopcua.github.io/

kohtala, 2026-09-16:
@Jellekon, I believe you better create a new issue. You can mention this issue on it, if you like to reference this for some reason.